PJLab-ADG/Livox-Mapping #1: livox lidar type and feature extract explanatory text

chenxianbo, 2022-01-04:
Qusention1：
    livox lidar type is horizon？
Qusention2:
    Have you understand feature extract detail？I miss in code，and I want apply it to Mid-70

zxd123, 2022-01-04:
Qusention1：
yes
Qusention2:
I don't know much about this. Its feature extraction is based on Lio livox. You can go to their home page.

SiyuanHuang95, 2022-01-04:
Hi chenxianbo,

Thanks for your interest in this repo.

Q1:  Yes. 
Q2: The feature extraction part is not our main focus. You may refer to the original one for more details. If you just need to apply this repo for one new sensor setting, I would suggest that you might need to check out the differences between Mid-70 and Horizon first.  Since we do not have Mid-70 available, no detailed help could be offered.

Bests