PJLab-ADG/Livox-Mapping #2: support other type lidar

qweesiic, 2022-02-21:
great job!!!
does it support other type lidar, such as velodyne\robosense and so on ?

SiyuanHuang95, 2022-02-22:
Hi, thanks for your interest!

To support other types of lidar, some modification works are needed:
1. You need one external IMU to publish the information. Also, extrinsic information is also needed.
2. If the point cloud pattern is different from Livox, the feature extraction block needs to de updated accordingly.
3.  The backend for LIOs with different types of lidar is similar.

Bests,

qweesiic, 2022-02-23:
> Hi, thanks for your interest!
>
> To support other types of lidar, some modification works are needed:
>
> 1. You need one external IMU to publish the information. Also, extrinsic information is also [...]

thank you for your reply!

closed 2022-02-23 by qweesiic