PJLab-ADG/Livox-Mapping #3: runtime error

WadeFrank, 2022-02-22:
roslaunch livox_odometry livox_odometry.launch save_path:="/home/qx263/livox_mapping/src/odom_path.txt"
... logging to /home/qx263/.ros/log/7909962c-93a2-11ec-9067-e4a7a0481322/roslaunch-qx263-28740.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://qx263:33043/

SUMMARY
========

PARAMETERS
 * /PoseEstimation/Extrinsic_Tlb: [1.0, 0.0, 0.0, -...
 * /PoseEstimation/IMU_Mode: 2
 * /PoseEstimation/filter_parameter_corner: 0.2
 * /PoseEstimation/filter_parameter_surf: 0.4
 * /PoseEstimation/save_path: /home/qx263/livox...
 * /ScanRegistration/config_file: /home/qx263/livox...
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /
    PoseEstimation (livox_odometry/PoseEstimation)
    ScanRegistration (livox_odometry/ScanRegistration)

auto-starting new master
process[master]: started with pid [28750]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7909962c-93a2-11ec-9067-e4a7a0481322
process[rosout-1]: started with pid [28763]
started core service [/rosout]
process[ScanRegistration-2]: started with pid [28774]
process[PoseEstimation-3]: started with pid [28781]
finished
[ScanRegistration-2] process has died [pid 28774, exit code -11, cmd /home/qx263/livox_mapping/devel/lib/livox_odometry/ScanRegistration __name:=ScanRegistration __log:=/home/qx263/.ros/log/7909962c-93a2-11ec-9067-e4a7a0481322/ScanRegistration-2.log].
log file: /home/qx263/.ros/log/7909962c-93a2-11ec-9067-e4a7a0481322/ScanRegistration-2*.log
[PoseEstimation-3] process has died [pid 28781, exit code -11, cmd /home/qx263/livox_mapping/devel/lib/livox_odometry/PoseEstimation __name:=PoseEstimation __log:=/home/qx263/.ros/log/7909962c-93a2-11ec-9067-e4a7a0481322/PoseEstimation-3.log].
log file: /home/qx263/.ros/log/7909962c-93a2-11ec-9067-e4a7a0481322/PoseEstimation-3*.log

When I play the rosbag, I meet error, I didn't know the reason. I think maybe I used the wrong version of Eigen. Which version of Eigen do you use? And how about the version of Ceres and GTSAM? 

Waiting for you reply, thank you!

SiyuanHuang95, 2022-02-22:
Hi, thanks for your interest!

we are using:
eigen：3.3.4
ceres：2.0.0.

However, I am fully recommending you just start from our provided docker, which will save you much time.

Bests,

SiyuanHuang95, 2022-02-28:
Since no further questions occurred, I will close this issue.

closed 2022-02-28 by SiyuanHuang95