PJLab-ADG/Livox-Mapping #4: livox_mapping node error

JinMing-Gu, 2022-02-27:
Hello, Thanks for great work!
Command: 
$ roslaunch livox_mapping livox_mapping.launch save_path:="/home/seven/Leraning/Livox-Mapping_note_ws/src/result"
$ roslaunch livox_odometry livox_odometry.launch save_path:="/home/seven/Leraning/Livox-Mapping_note_ws/src/result/pose.txt"
$ rosbag play my.bag

Problem:
livox_odometry works normally and livox_mapping has the following error. What are the possible reasons?
[image]

SiyuanHuang95, 2022-02-27:
Hi, thanks for your interest!
1. From your screen, no more information is available. Please provide your log file in the future.
2. One possible reason lies in the rosbag. If unindexed error occurs, please run "rosbag reindex yourbag.bag" at first, as http://wiki.ros.org/rosbag/Troubleshooting. Then roslauch again.

Bests,

SiyuanHuang95, 2022-02-27:
Hi, thanks for your file and reply.

1. Are you using your own collected rosbags? Have you tested with our provided rosbags?
2. The log file you provided is not informative, could you provide us with: livox_mapping-1*.log

[roslaunch][ERROR] 2022-02-27 14:17:14,613: [livox_mapping-1] process has died [pid 6422, exit code -11, cmd /home/seven/Leraning/Livox-Mapping_note_ws/devel/lib/livox_mapping/livox_mapping /aft_mapped_to_init:=/livox_odometry_mapped /velodyne_cloud_registered_local:=/livox_full_cloud_imu __name:=livox_mapping __log:=/home/seven/.ros/log/a96fe574-9793-11ec-b672-283a4d3b87ff/livox_mapping-1.log].
log file: /home/seven/.ros/log/a96fe574-9793-11ec-b672-283a4d3b87ff/**livox_mapping-1*.log**

Bests,

JinMing-Gu, 2022-03-07:
I think I have found a solution to the problem. Refer to this [github issue](https://github.com/gisbi-kim/FAST_LIO_SLAM/issues/8) and update the GTSAM version to the latest one.

SiyuanHuang95, 2022-03-07:
Cool, thanks for your information!

BTW, are you tested in the provided docker environment or your host pc?

JinMing-Gu, 2022-03-07:
> Cool, thanks for your information!
> 
> BTW, are you tested in the provided docker environment or your host pc?
I tested in my host pc, using the dataset you provided and the dataset I recorded myself can work normally.

SiyuanHuang95, 2022-03-07:
Okay. Glad to hear that the codes work finally.

I would recommend using docker for convenient usage. You can also upload your environment to the DockerHub for the community.

Since no new questions exist, I will close this issue. You can always re-open it when needed.

closed 2022-03-07 by SiyuanHuang95

JinMing-Gu, 2022-03-07:
> Okay. Glad to hear that the codes work finally.
> 
> I would recommend using docker for convenient usage. You can also upload your environment to the DockerHub for the community.
> 
> Since no new questions exist, I will close this issue. You can always re-open it when needed.

Thank you again for your excellent work！