PJLab-ADG/Livox-Mapping #5: catkin_make error

GeoArchofficiel, 2022-02-28:
First of all, thanks for your works and share it with all of us 🙂
i'm actually trying to catkin_make the Livox mapping but i have 2 make errors as shown in the pictures. can you help me to identify the problem and solve it?

glic@GLIC:~/catkin_ws$ catkin_make
Base path: /home/glic/catkin_ws
Source space: /home/glic/catkin_ws/src
Build space: /home/glic/catkin_ws/build
Devel space: /home/glic/catkin_ws/devel
Install space: /home/glic/catkin_ws/install
####
#### Running command: "cmake /home/glic/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/glic/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/glic/catkin_ws/install -G Unix Makefiles" in "/home/glic/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/glic/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2")
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/glic/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17")
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - ad_localization_msgs
-- ~~  - dnb_msgs
-- ~~  - camera_control_msgs
-- ~~  - pylon_camera
-- ~~  - livox_ros_driver
-- ~~  - livox_mapping
-- ~~  - livox_odometry
-- ~~  - xsens_mti_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ad_localization_msgs'
-- ==> add_subdirectory(Livox-Mapping/ad_localization_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ad_localization_msgs: 1 messages, 0 services
-- +++ processing catkin package: 'dnb_msgs'
-- ==> add_subdirectory(dragandbot_common/dnb_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- dnb_msgs: 4 messages, 4 services
-- +++ processing catkin package: 'camera_control_msgs'
-- ==> add_subdirectory(pylon-ros-camera/camera_control_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action camera_control_msgs/GrabAndSaveImage /home/glic/catkin_ws/src/pylon-ros-camera/camera_control_msgs/action/GrabAndSaveImage.action
-- Generating .msg files for action camera_control_msgs/GrabHDRImage /home/glic/catkin_ws/src/pylon-ros-camera/camera_control_msgs/action/GrabHDRImage.action
-- Generating .msg files for action camera_control_msgs/GrabImages /home/glic/catkin_ws/src/pylon-ros-camera/camera_control_msgs/action/GrabImages.action
-- camera_control_msgs: 22 messages, 14 services
-- +++ processing catkin package: 'pylon_camera'
-- ==> add_subdirectory(pylon-ros-camera/pylon_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'livox_ros_driver'
-- ==> add_subdirectory(livox_ros_driver)
-- livox_ros_driver version: 2.6.0
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.65.1
[... 602 lines cut ...]
[ 97%] Built target pylon_camera
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/glic/catkin_ws/src/Livox-Mapping/SC-PGO/src/laserPosegraphOptimization.cpp:19:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
[ 97%] Linking CXX executable /home/glic/catkin_ws/devel/lib/pylon_camera/IPConfigAuto
[ 97%] Built target IPConfigAuto
[ 97%] Linking CXX executable /home/glic/catkin_ws/devel/lib/livox_mapping/livox_mapping
/usr/bin/ld : ne peut trouver -lBoost::timer
collect2: error: ld returned 1 exit status
Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/build.make:479: recipe for target '/home/glic/catkin_ws/devel/lib/livox_mapping/livox_mapping' failed
make[2]: *** [/home/glic/catkin_ws/devel/lib/livox_mapping/livox_mapping] Error 1
CMakeFiles/Makefile2:7372: recipe for target 'Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/all' failed
make[1]: *** [Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/all] Error 2
[ 97%] Linking CXX executable /home/glic/catkin_ws/devel/lib/livox_odometry/PoseEstimation
[ 97%] Built target PoseEstimation
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

GeoArchofficiel, 2022-02-28:
/usr/bin/ld : ne peut trouver -lBoost::timer
collect2: error: ld returned 1 exit status
Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/build.make:479: recipe for target '/home/glic/catkin_ws/devel/lib/livox_mapping/livox_mapping' failed
make[2]: *** [/home/glic/catkin_ws/devel/lib/livox_mapping/livox_mapping] Error 1
CMakeFiles/Makefile2:7372: recipe for target 'Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/all' failed
make[1]: *** [Livox-Mapping/SC-PGO/CMakeFiles/livox_mapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

i think i resolved the error of sourcing livox_ros_driver but still have the error of -lboost::timer 
how can i solve it?

SiyuanHuang95, 2022-02-28:
You need to install iboost related.

Did you test on our docker?

GeoArchofficiel, 2022-02-28:
when i check the version of boost librery with "dpkg -s libboost-dev | grep 'Version'" i have this result "Version: 1.65.1.0ubuntu1"
sorry but i'm a beginner. how can i test on your docker?

SiyuanHuang95, 2022-02-28:
First of all, I cannot understand your terminal information with “/usr/bin/ld : ne peut trouver -lBoost::timer”. So no further help could be offer.

For the docker usage, please take a look at: https://github.com/PJLab-ADG/Livox-Mapping#23-docker

GeoArchofficiel, 2022-02-28:
if i use docker to run the test with docker , do i need to build the project with catkin_make?
thanks for your time.

SiyuanHuang95, 2022-03-01:
Yes, you need 

pull the docker -> get into the docker -> pull the project -> build the project

Bests

geoarch2021, 2022-03-03:
hi, i still have the same problem. can you provide me more informations or tutorial for docker build project please?
Thanks a lot

SiyuanHuang95, 2022-03-03:
Hi,

I would recommend you the official tutorial: https://docs.docker.com/get-started/

Since the docker part is not our main focus, I would suggest you go for help in the docker-related forum, where the people there are more professional than us.

Bests,
Siyuan

SiyuanHuang95, 2022-03-07:
Since no more active comments, I will close this issue. You can re-open it when needed.

BrightSoulXYHY, 2022-03-07:
I have the same problem

BrightSoulXYHY, 2022-03-07:
I fix it try add next line at `src\Livox-Mapping\SC-PGO\CMakeLists.txt`
it seems `livox_mapping` need `boost::timer` but compiler does not find it
```
# src\Livox-Mapping\SC-PGO\CMakeLists.txt
find_package(Boost REQUIRED COMPONENTS timer)
```

SiyuanHuang95, 2022-03-07:
Can you try it in our docker?

geoarch2021, 2022-03-07:
I installed docker and followed the instructions but seems it doesn’t work 😅
Someone can provide us a tutorial of the installation? Do we need to install Gtsam?

BrightSoulXYHY, 2022-03-08:
> Can you try it in our docker?

I using it on WLS and VM
I am unfamiliar to docker , I may learn docker if I have some leisure time next weekend hhhh

SiyuanHuang95, 2022-03-10:
> I installed docker and followed the instructions but seems it doesn’t work 😅 Someone can provide us a tutorial of the installation? Do we need to install Gtsam?

Hi @geoarch2021 , if you use the docker, I guess no more installation is needed since we have pre-installed everything we need in the docker environment.

For GTSAM, one possible reference would be: https://github.com/gisbi-kim/FAST_LIO_SLAM/issues/8.

For docker usage, please check the official guidance and tutorial.

Best

SiyuanHuang95, 2022-03-10:
> > Can you try it in our docker?
> 
> I using it on WSL and VM I am unfamiliar to docker , I may learn docker if I have some leisure time next weekend hhhh

@BrightSoulXYHY It definitely worth it using docker to try out the slam algorithms. 

Best

SiyuanHuang95, 2022-03-16:
Since no further questions, I will close this issue.

closed 2022-03-16 by SiyuanHuang95