PJLab-ADG/Livox-Mapping #6: replacing Livox_lio with FAST_LIO_2

sukhrajklair, 2022-03-03:
Hi 
Thank you for putting this together. 

We are using Mid-70 which we find working best with FAST_LIO_2. I was thinking of replacing LIO_LIVOX with FAST_LIO_2 for LIO and use a separate node to estimate ground plane (perhaps the one from hdl_graph_slam). However, I couldn't find where the ground constraint is added and how does SC_PGO receives the ground plane information (via topic?) Any pointers/hints would be much appreciated. 

Regards,
Sukhraj Klair

SiyuanHuang95, 2022-03-04:
Hi

Thanks for your interest in our project.

We add all ground-related constraints in LIO-Livox part.

For global constraint, refer to: https://github.com/PJLab-ADG/Livox-Mapping/blob/e896e6f0a09163eb2e8f22e63fee0024df03350e/LIO-Livox/src/lio/Estimator.cpp#L907

For single-frame parameter calculation, refer to: https://github.com/PJLab-ADG/Livox-Mapping/blob/e896e6f0a09163eb2e8f22e63fee0024df03350e/LIO-Livox/src/lio/PoseEstimation.cpp#L554

For the addition of constraint, refer to: https://github.com/PJLab-ADG/Livox-Mapping/blob/e896e6f0a09163eb2e8f22e63fee0024df03350e/LIO-Livox/src/lio/Estimator.cpp#L1020

For the specific constraint value calculation, refer to: https://github.com/PJLab-ADG/Livox-Mapping/blob/e896e6f0a09163eb2e8f22e63fee0024df03350e/LIO-Livox/include/utils/ceresfunc.h#L322

Bests,
Siyuan

sukhrajklair, 2022-03-04:
Ah I see. Thank you.

closed 2022-03-04 by sukhrajklair