PJLab-ADG/Livox-Mapping #8: maybe something wrong

chengwei0427, 2022-03-24:
Hi, @SiyuanHuang95 

I change the param **SLIDEWINDOWSIZE** from 2 to 3 in Estimator.h, and test the bag **casual_walk.bag** which lio_sam privode.

After lidar-imu inited, the trajectory drift .Used topics: /points_raw and /imu_raw. 
The extrinsics lidar -> IMU: 
 extrinsicTrans: [0.0, 0.0, 0.0]
  extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]

[image]

If set the SLIDEWINDOWSIZE=2, the program is normal.

[image]

The bag contains two imu topics(/imu_raw and /imu_correct). 

If used /imu_corrcect as the imu input, The extrinsics lidar -> IMU: 
 extrinsicTrans: [0.0, 0.0, 0.0]
  extrinsicRot: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]

Whether the **SLIDEWINDOWSIZE** is two or three, the program is normal.

It's strange, and maybe something wrong with the extrinsics params in the code.

Do you have a similar problem? Or any Suggestions?

SiyuanHuang95, 2022-03-26:
Have not met that issue. No further suggestions could be given. When you find anything interesting, you are quite welcome to share with us.

lhthq, 2022-03-26:
hi~ I see you issue.    In the terminal, there  are  “cannot create a KD-Tree with an empty input cloud”？ Why?

chengwei0427, 2022-03-28:
> hi~ I see you issue. In the terminal, there are “cannot create a KD-Tree with an empty input cloud”？ Why?

There are three KD-trees, corresponding to the three types of features(edge, surf and non), you could check the three types of feature, maybe one of them is empty.

chengwei0427, 2022-04-01:
[https://github.com/TixiaoShan/LIO-SAM/issues/204](url)

SiyuanHuang95, 2022-04-01:
> [https://github.com/TixiaoShan/LIO-SAM/issues/204](url)

Cannot catch your main idea, was the transformation matrix will cost this issue?

Real url:

[https://github.com/TixiaoShan/LIO-SAM/issues/204](https://github.com/PJLab-ADG/Livox-Mapping/issues/url)

SiyuanHuang95, 2022-04-12:
Since there is no further update, I will close this issue.

closed 2022-04-12 by SiyuanHuang95