PJLab-ADG/Livox-Mapping #10: the number of the pcd and odom files generated by Interactive_slam packages is not enough

yijing2050, 2022-03-25:
Hello! When I sue ctrl+c to shut down the odometry and mapping thread after the "around_park_2021-09-24-11-30-54.bag" plays over, only one odom file is generated. There are only 5.9MB of the odom and pcd files generated by the odometry_saver package even though I usde ctrl+c before the bag play over. With the help of zxd,one fomer collegue of your lab, we finally haven't solved the problem after trying many times. And I make the sc files using them, launch the livox_localization, it couldn't match the index. But I use the https://github.com/SMRT-AIST/interactive_slam/wiki/Example2, the number of the pcd and odom files is enough. 
Looking forwards to your reply.Thanks.
the generated pcd and odom files is as the image
[image]

SiyuanHuang95, 2022-03-26:
Fixed. 

Two things to the community:
1. We are open-sourcing for research & academic usage, commercial purposes are not encouraged. And we have not tested for large-scale usage.
2. The problem is due to the environmental issue.

closed 2022-03-26 by SiyuanHuang95

yijing2050, 2022-03-26:
Thanks for your reply although what you said don't work for the issue. I have found the reason and solved the question.@[SiyuanHuang95](https://github.com/SiyuanHuang95)

> Fixed.
> 
> Two things to the community:
> 
>     1. We are open-sourcing for research & academic usage, commercial purposes are not encouraged. And we have not tested for large-scale usage.
> 
>     2. The problem is due to the environmental issue.

SiyuanHuang95, 2022-03-26:
What I said is something more general but not related to this issue. 

It's not a gentleman-style to contact the developer directly with IM and to apply this research-purpose repo to somewhere else.