PJLab-ADG/Livox-Mapping #12: use for livox mid70

gongyue666, 2022-03-28:
Hi I am using livox mid70 and a extern IMU. the problem is that lidarFeatureExtractor node always output logs: too few ground points. and the other nodes seem run nothing. I have change the horizon_config.yaml to mid70.yaml like this:

`%YAML:1.0

# switches
Lidar_Type: 0    # 0-horizon
Used_Line: 1    # lines used for lio, set to 1~6
Feature_Mode: 0    # 0(false) or 1(true)
NumCurvSize: 2
DistanceFaraway: 100 # [m]  <DistanceFaraway near / >DistanceFaraway far
NumFlat: 3 # nums of one part's flat feature
PartNum: 150 # nums of one scan's parts
FlatThreshold: 0.02 # cloud curvature threshold of flat feature
BreakCornerDis: 1 # break distance of break points
LidarNearestDis: 0.50 # if(depth < LidarNearestDis) do not use this point 
KdTreeCornerOutlierDis: 0.2 # corner filter threshold
Use_seg: 1 # use segment algorithm
map_skip_frame: 2`

how can I modify the params to make the project can work  with mid 70?

SiyuanHuang95, 2022-03-30:
Too few ground points are the result of feature extraction. Could you provide some screenshots?

SiyuanHuang95, 2022-03-31:
Thanks for your sharing. 

Currently, no much time to test your bag. Will try it in the near future.

gongyue666, 2022-04-01:
ok，thx all the same，I will try to slove it and share progress

SiyuanHuang95, 2022-04-12:
Since there is no further update, I will close this issue.

closed 2022-04-12 by SiyuanHuang95