PJLab-ADG/Livox-Mapping #13: RTK factor

WadeFrank, 2022-03-28:
`                mtxPosegraph.lock();
                {
                    // odom factor
                    gtSAMgraph.add(gtsam::BetweenFactor<gtsam::Pose3>(prev_node_idx, curr_node_idx, poseFrom.between(poseTo), odomNoise));

                    // // rtk factor 
                    // if(hasRTKforThisKF) {
                    //     gtsam::Pose3 poseRTK = Pose6DtoGTSAMPose3(currRTK);
                    //     gtSAMgraph.add(gtsam::PriorFactor<gtsam::Pose3>(curr_node_idx, poseRTK, RTKNoise));
                    //     cout << "RTK factor added at node " << curr_node_idx << endl;
                    // }
                    initialEstimate.insert(curr_node_idx, poseTo);                
                    // runISAM2opt();
                }
                mtxPosegraph.unlock();
`

我看到你在上面代码中将因子优化的rtk部分给注释掉了，请问是什么原因呢？

SiyuanHuang95, 2022-03-30:
原因：Gnss标定的不好，而且园区内信号一般，加里面融合后结果反而更差，就去掉了

如果需要的话，去掉注释，统一GNSS-topic

WadeFrank, 2022-03-30:
好的，非常感谢！