PJLab-ADG/Livox-Mapping #16: Parameters between Lidar and Imu/GNSS

ManChrys, 2022-07-19:
Hello,

i have a livox mid 40 and an INS/GNSS system (xsens mti g 710) . I would like to ask you whats these parameters mean

> 
    <rosparam param="Extrinsic_T_livox2gnss"> [0.997573, -0.024254, 0.065274, 0.322019,
                                               0.023477, 0.999644, 0.012663, 0.043158,
                                               -0.066402, -0.011080, 0.997733, 0.456606,
                                               0.0, 0.0, 0.0, 1.0]</rosparam>

I understand that some of them are the distances between lidar and imu but could you explain every number on this table what represents?

Thanks in advance.

SiyuanHuang95, 2022-07-24:
Hi,

Thanks for your interest in our project.

The T matrix contains the Rotation matrix and translation vector.

R = T [0:2, 0:2] 
t = T[0:2, 3]

so you would figure out that the rotation matrix is almost an identity matrix and translation vector is small.

Best

InterKnight, 2022-08-04:
> Hi,
> 
> Thanks for your interest in our project.
> 
> The T matrix contains the Rotation matrix and translation vector.
> 
> R = T [0:2, 0:2] t = T[0:2, 3]
> 
> so you would figure out that the rotation matrix is almost an identity matrix and translation vector is small.
> 
> Best

[0.997573, -0.024254, 0.065274, 0.322019,
 0.023477, 0.999644, 0.012663, 0.043158,
-0.066402, -0.011080, 0.997733, 0.456606,
 0.0, 0.0, 0.0, 1.0]

that means 0.322019 is x translation， 0.043158 is y translation, 0.456606 is z translation ?

InterKnight, 2022-08-04:
Hi，think you for great project.
I have another question about parameters:
why parameters in livox_odometry.launch is not same with that in livox_mapping.launch ?

SiyuanHuang95, 2022-08-04:
> [0.997573, -0.024254, 0.065274, 0.322019, 0.023477, 0.999644, 0.012663, 0.043158, -0.066402, -0.011080, 0.997733, 0.456606, 0.0, 0.0, 0.0, 1.0]
>
> that means 0.322019 is x translation， 0.043158 is y [...]

@InterKnight Right

SiyuanHuang95, 2022-08-04:
> Hi，think you for great project. I have another question about parameters: why parameters in livox_odometry.launch is not same with that in livox_mapping.launch ?

@InterKnight Usually the translation part does not affect the final result much. And the rotation part is almost identical.

InterKnight, 2022-08-05:
> > Hi，think you for great project. I have another question about parameters: why parameters in livox_odometry.launch is not same with that in livox_mapping.launch ?
> 
> @InterKnight Usually the translation part does not affect the final result much. And the rotation part is almost identical.

thank you very much!

SiyuanHuang95, 2022-08-05:
When no further questions, I will close this issue.

closed 2022-08-05 by SiyuanHuang95