PJLab-ADG/Livox-Mapping #19: Can not build follow your steps

wdwz2000, 2022-09-22:
1. Can not pass the step 3-Build  .

When I execute the sentence :_git clone https://github.com/PJLab-ADG/LIO_Livox_SLAM.git_, 
system responds me :_remote: Repository not found._
Is the website still be right?

2. Besides, step 2.3-Docker :_docker run --gpus all -it -p 3316:22 --device=/dev/dri --group-add video --volume=/tmp/.X11-unix:/tmp/.X11-unix  --env="DISPLAY=$DISPLAY" siyuanhuang95/livox_slam:release /bin/bash_, I can not run, either , what does it mean? Is it necessary to execute? 
system responds me:_docker: Error response from daemon: error gathering device information while adding custom device "/dev/dri": no such file or directory._

SiyuanHuang95, 2022-09-28:
Hi
1. Thanks for pointing out our address error, now fixed.
2. It's the command for docker env, you can create that folder or remove that parameter.

wdwz2000, 2022-09-28:
Thank you for your reply
But I don't understand the command` docker run --device=/dev/dri `very well
What directory is this /dev/dri? Is it the livox ladar mount directory? Or what device directory?(my device is livox-mid-70)

I am Chinese and my English is not very good. Can we talk in Chinese?

感谢您的回复
但是我对于docker run --device=/dev/dri这个命令不是很理解
请问这个/dev/dri是什么目录？是livox雷达挂载目录吗？还是什么设备的目录？（我的设备是 livox——mid-70）
最后，不好意思，我是个中国人，英语不太好，请问我们能不能用中文交谈？

SiyuanHuang95, 2022-09-29:
Hi,

--device declares the folder where your docker could operate. It has no relation with Livox. 

这个目录只是docker相关的设定，和livox硬件无关

wdwz2000, 2022-09-29:
所以这条命令不需要我依据实际情况进行更改是吗？可是在我执行之后会给我报错：
`docker: Error response from daemon: error gathering device information while adding custom device "/dev/dri": no such file or directory.`
如果我手动创建这个文件夹，再执行这条命令，会报错：
`docker: Error response from daemon: error gathering device information while adding custom device "/dev/dri": not a device node.`
请问我该怎么办？（板子是nvidia tx2，arm64架构）

SiyuanHuang95, 2022-09-29:
I guess the biggest problem is from the OS.

/dev/dri usually contains the graphical related, like Nvidia/xlog, etc. We use this command to let the docker use host-pc's display.

We have not tested it with TX2, so no further help could be offered.

SiyuanHuang95, 2022-10-03:
Since there is no further update, I will close this issue.

closed 2022-10-03 by SiyuanHuang95