PJLab-ADG/Livox-Mapping #21: 缺失头文件NavStateInfo.h

wdwz2000, 2022-11-05:
fatal error: ad_localization_msgs/NavStateInfo.h: 没有那个文件或目录
 #include <ad_localization_msgs/NavStateInfo.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
请问这个头文件是不是您不小心误删啦，麻烦重新上传一下呗，谢谢啦

SiyuanHuang95, 2022-11-09:
The header file is generated with the ros msg file, from: https://github.com/PJLab-ADG/Livox-Mapping/blob/main/ad_localization_msgs/msg/NavStateInfo.msg.

You can try it another time since the compelling did do some tricky work.

wdwz2000, 2022-11-09:
收到,我再试试, 感谢感谢

wdwz2000, 2022-11-09:
顺便问一下, 我的tx2不能使用您的docker, 是不是没有什么影响?只要我自己把tx2环境配置好是不是就行了

SiyuanHuang95, 2022-11-10:
> 顺便问一下, 我的tx2不能使用您的docker, 是不是没有什么影响?只要我自己把tx2环境配置好是不是就行了

Have not tried TX2, and no help and ideas could offer.

And you are welcome to give the update when you finally make it! :-)

closed 2022-11-11 by SiyuanHuang95