PJLab-ADG/Livox-Mapping #23: livox 的FOV小，使用SCANCONTEXT 这类全局描述子时，应该怎么处理呢？

kahowang, 2022-11-21:
Hi,thanks for your great work! I've got a question about whether this global description can work for livox-like lidar.As far as I concerned,scan Context describe the height information of 360. But solid lidar like livox-mid 70 has too small fov of 70,and its pattern is non-repeative .Can scanContext work to computer the similarity of two frame of livox-like lidar? Thank you sincerely!

SiyuanHuang95, 2022-11-21:
Hi, thanks for your interest in our project!

Correct, the FoV issue did make global localization one big problem.  For normal routine, it still works most time but would fail when through the inverse routine.  The non-repetitive pattern does not affect too much in our experiment.

Bests

kahowang, 2022-11-22:
Thank you very much~

closed 2022-11-22 by kahowang