PJLab-ADG/Livox-Mapping #29: mid360 support

dbaldzhiev, 2023-05-02:
LIO-Livox now supports Mid360 sensors. Do you plan to reintegrate with the updated version of LIO-Livox.
I am trying to do it myself https://github.com/dbaldzhiev/Livox-Mapping but with no success. 
My fork compiles fine but then :
`[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 0!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[PoseEstimation-3] process has died [pid 16720, exit code -11, cmd /root/ws_livox_mapping/devel/lib/livox_odometry/PoseEstimation __name:=PoseEstimation __log:=/root/.ros/log/e1984e1e-e8be-11ed-84b0-0242ac110002/PoseEstimation-3.log].
log file: /root/.ros/log/e1984e1e-e8be-11ed-84b0-0242ac110002/PoseEstimation-3*.log`

SiyuanHuang95, 2023-05-06:
1. Currently, no.
2. Please check your input point cloud status, like whether it is empty

closed 2023-05-16 by SiyuanHuang95

yueyihua, 2024-05-27:
When I used the MID-360, the following warning message appeared：
[ WARN] [1716780577.472366015]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame aft_pgo (parent camera_init) at time 1714013221.395308 according to authority unknown_publisher

SiyuanHuang95, 2024-05-27:
It seems to be due to the inconsistent frame base name? Since in this repo, we hard-coded some famebase names